Team3176/Code_2024 #19: Investigate why Blinkin stopped working.  Begin by HW people taking apart and taking photos of PCB, looking for discolorations or deformations like melting.  Possibly consider beginning to develop our own LED library like the Blinken one that uses addressibleLEDs directly instead of the Blinkin.

char3176, 2024-03-07:


SamuelWillman, 2024-03-07:
There may already be an adressableLED library on one of the laptops.

SamuelWillman, 2024-03-07:
At least, we started it on the laptops, I dont know if we actually kept it or not.

SamuelWillman, 2024-03-08:
There are people using the robot, when we can acess it ourselves, we will have to factory reset the blinkin and hope that it helps.

char3176, 2024-03-08:
While using the robot later that evening, the leds suddenly came back.  We are clearly experiencing problems that others on ChiefDelphi have also reported.

@SamuelWillman I believe we should proceed to develop our own library using addressibleLEDs that replicates the Blinkin capabilities.

Does anyone object?

char3176, 2024-03-08:
I will wait for @SamuelWillman's input, but at present I think we are ready to close this "issue" and open a new one that involves building a library that replicates the Blinkin pattens.  

Any last input/comments @SamuelWillman, or are we good to close this "issue"?  And if we are good to close, I'm going to walk you thru how to do that at Fri meeting.

char3176, 2024-03-12:
I'm creating a new issue for development of a library using addressibleLEDs that replicates the Blinkin functionality.

char3176, 2024-03-12:
#41

char3176, 2024-03-12:
#41

closed 2024-03-12 by char3176