Team3176/Code_2024 #26: Implement PID on climb in roll on just 1 arm using NavX roll value.

char3176, 2024-03-08:


jheidegger, 2024-03-08:
Take a look in drivetrain there are the functions
getChassisRoll()
getChassisPitch()

That can be useful for getting the values

char3176, 2024-03-14:
@222003791 was able to get motor linked to gyro value.  However appears to be continuous, ie not isolated to running only when xbox button pressed.  That is next problem to solve.  After that, we need to make sure motor is going in opposite directions when board tilted in opposite directions.  Once these are resolved, we are ready to attempt to do it with 2 arms.

char3176, 2024-03-16:
Was continuous b/c was being called in @periodic.  Still experiencing issues with linking to xbox button.

char3176, 2024-03-20:
Climb on 1 arm completed.  Prototype for Climb on 2 arm is in PR 65

closed 2024-03-20 by char3176