Team3176/Code_2024 #29: Fix three cameras with broken JST wires.

char3176, 2024-03-08:


WolfeeZed, 2024-03-09:
Approx 3.6 mm height (Outside)
Approx 7.5 mm width (outside)
Approx 6.5mm width (inside) 
Approx 3 mm length (inside)
[image]

Do not have the correct size of JST housing, we need more ordered

char3176, 2024-03-09:
I think it might be a JST-ZH 4 pin female connector.  I've placed order for them.  Should have in time for Monday meeting to try.

jheidegger, 2024-03-10:
https://www.chiefdelphi.com/t/frc-6328-mechanical-advantage-2024-build-thread/442736/151?u=jheidegger
I think Chris is right as described in this thread

char3176, 2024-03-11:
Yeah, they arrived and they fit perfect.  However they came with precrimp leads, so we'll need to splice those onto the existing wire and heatshrink, Braeden.  Can do Mon night if you are at meeting.

WolfeeZed, 2024-03-11:
Yep, I’ll be there tonight and I can do it right after I’m done reconnecting the wires on the practice defense bot to the PDH

WolfeeZed, 2024-03-11:
All 3 camera wires have been fixed and cases have been added to them for future use.

closed 2024-03-11 by WolfeeZed