Team3176/Code_2024 #36: Investigate possiblity of a operator button that can turn a driver cam feed in Elastic on and off for intaking.  Not sure it's possible tho.

char3176, 2024-03-09:


Marbledata, 2024-03-11:
ive looked into this, and the chances arent looking good to me, I think that the setDriverMode() method might be able to do this, we could probably tie it to a button. Problem is that no matter what I cannot actually get a camera feed in elastic, that needs to be solved before setDriverMode() can be tested

char3176, 2024-03-12:
Liam, take a look at this WPILib page and see if we can use it's method to publish the feed from the NoteCam OrangePi to Elastic.  

https://docs.wpilib.org/en/stable/docs/software/vision-processing/introduction/cameraserver-class.html

jheidegger, 2024-03-12:
I also think these pages will be useful. https://docs.wpilib.org/en/stable/docs/software/dashboards/shuffleboard/layouts-with-code/index.html
Since we will want to add and possibly remove the camera widget to reduce the bandwidth

jheidegger, 2024-03-12:
Elastic by default does not populate widgets (unlike shuffle board which does). You can add a camera stream to elastic from the CameraPublisher NT4 topic. Here is an example from simulation.
[image]

char3176, 2024-03-12:
Liam can you try what Jon suggests above?  That would be a start to at least getting the streams on Elastic.  We can deal with the next step of turning it on and off after you get it streaming.

jheidegger, 2024-03-13:
Ok so I poked around a bit and its actually pretty simple. We just need to change the streams [] NT4 field. 

- So we should create our own camera publisher that is pointed at a bogus URL possibly using the SendableCameraWrapper
- then create a NT4 subscriber to get the stream url from the PhotonCamera we want
- then use a NT4 publisher to overwrite the streams [] field of the SendableCameraWrapper. 

Example:
NetworkTableInstance.getDefault()
            .getTable("CameraPublisher/camera1")
            .getStringArrayTopic("streams")
            .publish();

char3176, 2024-03-14:
@Marbledata can we test Jon's hypothesis above on Thur?  Looks promising.

Marbledata, 2024-03-14:
Will work on it today

Marbledata, 2024-03-15:
[image]
I know that I need to use a NT4 publisher to overwrite the streams value of the sendablecamerawrapper, I just have no idea how to do that. even if i get a publisher, I dont know how to get it to change the streams value

char3176, 2024-03-19:
Besides the bad choice of putting the switching button on transStick button 1, this looks ok.  

@Marbledata can you take a control board with an attached photonvision unit and camera, verify functionality and then measure and report here the max perfmon values, please.