Team3176/Code_2024 #66: Conveyor Tuning Speeds

jheidegger, 2024-03-20:
"Intake process would as follows:
-4 Bar down and intake spinning
-Conveyor spinning at speedA
-Note triggers LaserCAN1
-Intake returns and Conveyor slows to SpeedB
-Note Triggers LaserCAN2
-Conveyor stops (edited) " - from Ben

char3176, 2024-03-20:
@jheidegger Logic has mildly changed.  LzrCAN2 is now "in front" of LzrCAN1, therefore LzrCAN2 has to trip twice (once for each front of note, and second for back of note) before stopping conveyor.

jheidegger, 2024-03-21:
reminder to check direction and invert motor config if needed. Positive is note going towards shooter, negative is note going out towards intake

jheidegger, 2024-03-21:
isLaser2 and isLaser1 are exposed for the intakeNote logic in superstructure

jheidegger, 2024-04-10:
Conveyor logic is implemented with linebreaks and worked at wash and state

closed 2024-04-10 by jheidegger